sxjack/uav_electronic_ids #3: Have you considered a Wi-Fi beacon implementation

gabrielcox, 2021-04-13:
Wi-Fi beacon (as specified in the ASD-STAN doc) should be a merging of what you already did for France (information element 0xdd (221)), but encode the data as it is in OpenDroneID. 

This is all real good work!

Gabriel.

sxjack, 2021-04-13:
I looked at it when I installed version 2.1.1 of the receiver app on my phone, but I have no way of testing it as my phone doesn't support it.

I am currently waiting to see what the UK government is going re remote identification. I was expecting them adopt EN 4709-002, but last I heard they haven't made their mind up.

gabrielcox, 2021-04-13:
Parrot just updated the list of Beacon tested phones:

https://github.com/opendroneid/receiver-android/commit/6d0dd4413bc072845c75f8c411018ee0f5c4380d

sxjack, 2021-04-13:
I don't need a new phone and if I did, its ability to receive 4709-002 remote ids would be way down my required feature list.

If somebody can point me at a way of generating known good beacon signals (without buying a Parrot drone), then I would add beacon support to my ESP32 scanner and then I could do an ESP32 beacon class.

sxjack, 2021-04-27:
I have added ODID beacons to my id_open library and scanner. 

Frames generated by the library are read by the scanner, but I have no way of testing against known good apps or devices.

gabrielcox, 2021-04-28:
> I have added ODID beacons to my id_open library and scanner.
> Frames generated by the library are read by the scanner, but I have no way of testing against known good apps or devices.

Awesome!  I will build/run and test it against a Galaxy S10e w/Android ODID Receiver and let you know of the results.

gabrielcox, 2021-05-03:
I was able to get it to work with the ODID Android Receiver after putting the minor (temp) changes that are in my fork.  I was having problems with the 221 IE declared size (253 bytes) not matching the size being transmitted (35 bytes).  So I did a temp hard code just to get it working -- that will need to be re-done.
Also, I added the required App Code and Counter Bytes (counter is not yet working, but the offsets are now correct).  We are going to change the color codes of those in the specification to make it more clear that those need to be set. 

[image]

Also, as you can see, the current OUI in the ASD STAN spec is "placeholder" where Parrot was using their own.  ASD-STAN has now acquired their own OUI (FA-0B-BC), and the plan is to use this going forward.  ( I haven't made this change yet in the fork or in the Android receiver.  That's likely coming next.
The OUI list is [here](http://standards-oui.ieee.org/cid/cid.txt).

sxjack, 2021-05-04:
I have updated id_open and id_scanner.

Length should be OK, counter added, #define for selecting OUI.

I don't have a copy of the standard :-).

gabrielcox, 2021-05-05:
Will do more testing and let you know.

gabrielcox, 2021-05-09:
For Reference:
[image]

gabrielcox, 2021-05-24:
I have tested this implementation with the receiver app and in the process, found an issue with the receiver app crashing with Wi-Fi Beacons.  It is now fixed as of 2.2.1.  I created a PR to combine messages into a message pack as the message pack is intended.  It was only a minor change of setting the appropriate fields as "valid" to do this.
It now works quite nicely!

An Open Drone ID Wireshark dissector would be handy for validation, so I created it here:
[https://github.com/opendroneid/wireshark-dissector](https://github.com/opendroneid/wireshark-dissector)

[image]

sxjack, 2021-05-26:
I can see the frames on Wireshark, but I can't get the dissector to work which doesn't surprise me as, as far as I am concerned, Wireshark is not particularly user friendly and I am running it on an old netbook.

sxjack, 2021-05-27:
I can see the ODID beacon frames, it is just that I am seeing them raw rather than decoded. My ESP32 RID scanner picks them up and decodes them OK.

gabrielcox, 2021-05-27:
Ok, good.  Did you double check the .lua file is in the "Personal Lua Plugins" folder?  Also, you can re-load/re-run  what's in that directory using CTRL+SHIFT+L.