sxjack/uav_electronic_ids #4: Debug output

gabrielcox, 2021-05-03:
Are you able to get the debug output to the serial port when DIAGNOSTICS is set to 1 ?  I'm not seeing anything.  I'm about to see the other "Arduino" serial debug output, but not from C.

sxjack, 2021-05-04:
DIAGNOSTICS only outputs anything for bluetooth. It requires the serial port to be opened in the parent application.

sxjack, 2021-05-05:
Debug output added for beacons.

ID_OpenDrone::transmit_wifi 187 67+28=95 * 80 ... 37 39 52 | dd 23 90 3a e6 0d 4c | f0 19 01 20 a0 01 ... 00
ID_OpenDrone::transmit_wifi 187 67+28=95 * 80 ... 37 39 52 | dd 23 90 3a e6 0d 4d | f0 19 01 10 02 38 ... 00
ID_OpenDrone::transmit_wifi 187 67+28=95 * 80 ... 37 39 52 | dd 23 90 3a e6 0d 4e | f0 19 01 00 20 47 ... 00
ID_OpenDrone::transmit_wifi 187 67+28=95 * 80 ... 37 39 52 | dd 23 90 3a e6 0d 4f | f0 19 01 10 02 38 ... 00

gabrielcox, 2021-05-05:
Ah -- of course, I totally missed that I wasn't even hitting those sections. :-D .
This is a good layout of debug (Frame Header | VSIE Header | ODID Message) 👍 .

closed 2021-05-05 by gabrielcox