sxjack/uav_electronic_ids #6: 'CLOCK_MONOTONIC' Error on compile

Tissy, 2021-07-26:
Trying to compile on an Arduino ESP32 and I am getting this error, any suggestions to resolve please?

Many thanks
```
D:\Users\Steven\Documents\Arduino\libraries\libopendroneid\wifi.c: In function 'odid_wifi_build_nan_sync_beacon_frame':
D:\Users\Steven\Documents\Arduino\libraries\libopendroneid\wifi.c:237:16: error: 'CLOCK_MONOTONIC' undeclared (first use in this function)
  clock_gettime(CLOCK_MONOTONIC, &ts);
                ^
D:\Users\Steven\Documents\Arduino\libraries\libopendroneid\wifi.c:237:16: note: each undeclared identifier is reported only once for each function it appears in
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Steven\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: D:\Program Files (x86)\Arduino\libraries\WiFi
Using library id_open at version 0.8 in folder: D:\Users\Steven\Documents\Arduino\libraries\id_open 
Using library BLE at version 1.0.1 in folder: C:\Users\Steven\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\BLE 
Using library utm at version 1.1 in folder: D:\Users\Steven\Documents\Arduino\libraries\utm 
Using library libopendroneid in folder: D:\Users\Steven\Documents\Arduino\libraries\libopendroneid (legacy)
Using library WiFi at version 1.0 in folder: C:\Users\Steven\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi 
exit status 1
Error compiling for board ESP32 Wrover Module.

```

sxjack, 2021-07-26:
libopendroneid is not an arduino library. 

Put opendroneid.c, opendroneid.h and odid_wifi.h in libraries/id_open. The version of wifi.c in id_open is modified to remove this error.

Tissy, 2021-07-27:
That's great, thank you for your help.

closed 2021-07-31 by sxjack