sxjack/uav_electronic_ids #7: ESP32 Differences

Tissy, 2021-08-09:
Hi,

If I use an ESP32 (Lilygo TTGO T1_v1.3 - https://github.com/LilyGO/ESP32-TTGO-T1) I get a steady stream of data from the jSON output on USB as expected (although with no display attached at the moment).

However, if I use a different ESP32 module with TFT display (Lilygo TTGO T4_v1.3 - https://www.aliexpress.com/item/32854502767.html) I only get a jSON output every 60 seconds and all fields are essentially zero.  

`23:02:49.685 -> { "index": 8, "runtime": 780, "mac": "00:00:00:00:00:00", "id": "NONE", "uav latitude":    0.000000, "uav longitude":    0.000000, "alitude msl": 0, "height agl": 0, "base latitude":    0.000000, "base longitude":    0.000000, "speed": 0, "heading": 0 }`

Although the display perhaps isn't configured just yet, I would expect to see an output similar to the  TTGO T1_v1.3 module.

Any idea on where to start looking at the issue please?

Thank you for your help.

sxjack, 2021-08-10:
id_scanner outputs that line every 60 seconds if it hasn't received any data to show that it is still alive.

What are you using to transmit ID's for id_scanner to receive?

Tissy, 2021-08-12:
Initially I was using another ESP32 to generate the IDs, however I am getting the expect output on an Anafi.

sxjack, 2021-08-13:
Interesting, but not particularly surprising, to hear that a Parrot Anafi with UK firmware is transmitting an ID.

Tissy, 2021-11-10:
Hi Steve,

I've noticed a couple of differences.

If using an ESP32 to generate a RemoteID or indeed random flight, the scanner picks this up fine with...

`esp_wifi_set_channel(6,WIFI_SECOND_CHAN_NONE);`

However, when using a Parrot ANAFI which is known to transmit a remote ID, the scanner doesn't pick up the telemetry from the aircraft unless its changed back to...

`  esp_wifi_set_channel(6,(wifi_second_chan_t) 10);`

But then the ESP transmitter stops being received.

Any suggestions so the scanner can detect both ESP transmitter and a Parrot UAS as they are both on the same channels.

sxjack, 2021-11-14:
I wish that ASTM/ASD-STAN/opendroneid etc had just gone with Bluetooth 4 :-). 

I only set the second channel because I was trying to get the opendroneid sender program working and my home network was on channel 10 and I didn't want to change it. IIRC the ESP32 didn't pick up the signals until I moved my home network to channel 6. There seems to be something strange about how esp_wifi_set_channel() works.

Is your ESP32 transmitter transmitting NAN or beacon?

Can you confirm that that Anafi is transmitting on channel 6?

Tissy, 2021-11-14:
Indeed, would make things a little easier :-)

I couldn't see it in the code, where did you set the transmitting channel for the opendroneid, I shall have a play.  The ESP32 is transmitting a beacon.

Yes, I believe the Anafi is transmitting it's RemoteID on channel 6, but the scanner only picks it up with this configuration:

`esp_wifi_set_channel(6,(wifi_second_chan_t) 10);`

I presume it's setting the wifi_second_channel to 10?

friissoren, 2021-11-15:
Just FYI: Both the ASTM and the ASD-STAN specifications state for Wi-Fi Beacon that you can either use any 2.4 GHz or 5 GHz channel you like but then have to transmit at least 5 times a second or you can use either Channel 6 or 149 and then only transmit 1 time per second.

(NaN transmit is a bit simpler, since the NaN spec says that only channel 6 and 149 are allowed. But although in Europe NaN is a valid option, the current draft Means of Compliance statement from ASTM for the US, mandates that only Wi-Fi Beacon [can be used](https://github.com/opendroneid/opendroneid-core-c#relevant-specifications), if using Wi-Fi).

@CaliCaliIT: Is it possible you could share the changes you came up with to improve the problems you saw in issue #9? I somehow get the feeling that it might be partly related to the problems reported in this issue?

sxjack, 2021-11-15:
> Both the ASTM and the ASD-STAN specifications state for Wi-Fi Beacon that you can either use any 2.4 GHz or 5 GHz channel you like but then have to transmit at least 5 times a second or you can use either Channel 6 or 149 and then only transmit 1 time per second.

I was going to ask you this. I only have the original ASTM standard which says channel 6 for NAN and the French beacon specification which also says channel 6.

So NAN is on the way out. I can't say that I am surprised.

sxjack, 2021-11-15:
> I couldn't see it in the code, where did you set the transmitting channel for the opendroneid, I shall have a play.

It's near the beginning of the constructor. As it looks like people might want to change it I will make it a #define. 

> I presume it's setting the wifi_second_channel to 10?

That is what it is supposed to do. I have my doubts.

I am currently changing the code to use the new odid_wifi_build_message_pack_beacon_frame() function and remove some old, redundant code, but I doubt if this will make any difference.

sxjack, 2021-11-15:
I've added an option to dump the ODID beacon frame to the serial output to id_scanner.

closed 2022-05-27 by sxjack