sxjack/uav_electronic_ids #10: sending and receiving ODID data simultaniously

Mxmaxmx, 2021-11-03:
I have been trying to put together the Id_scanner and Id_open codes to both receive and transmit data simultaneously mostly via Wifi-NAN. They seem to have different wifi settings for either rx or tx. Is it a way to make them work simultaneously on the same ESP32 board or is it doomed to fail?

sxjack, 2021-11-03:
I don't know. I have never looked at doing this. I wrote the scanner because I don't have an Android device that will receive NAN and don't have an app that will receive French style beacon frames.

I don't like the way that the id_open library sets up the ESP32 Wi-Fi and have been meaning to see if there is a better way to do it.

Mxmaxmx, 2021-11-03:
If it gets set up in a way that works with the promiscuity setting for the scanner, then it might work. Because I´m getting readings from other nodes now but get an error message every time squitter.transmit executes.

Mxmaxmx, 2021-11-04:
There seem to be a way to send on 802.11 with esp_wifi_80211_tx() and that is while promiscuity is running.

closed 2022-05-27 by sxjack