sxjack/uav_electronic_ids #11: Compiling id_open

pierreSCpapon, 2021-11-04:
When compiling. I encountered the following error:

`lib\libopendroneid\wifi.c: In function 'odid_wifi_build_nan_sync_beacon_frame':
lib\libopendroneid\wifi.c:261:11: error: 'struct ieee80211_beacon' has no member named 'element_id'
     beacon->element_id = 0xDD;
           ^
lib\libopendroneid\wifi.c:262:11: error: 'struct ieee80211_beacon' has no member named 'length'
     beacon->length = 0x22;
           ^
lib\libopendroneid\wifi.c:263:18: error: 'struct ieee80211_beacon' has no member named 'oui'
     memcpy(beacon->oui, wifi_alliance_oui, sizeof(beacon->oui));
                  ^
lib\libopendroneid\wifi.c:263:57: error: 'struct ieee80211_beacon' has no member named 'oui'
     memcpy(beacon->oui, wifi_alliance_oui, sizeof(beacon->oui));
                                                         ^
lib\libopendroneid\wifi.c:264:11: error: 'struct ieee80211_beacon' has no member named 'oui_type'
Compiling .pio\build\m5stack-core2\lib6c7\id_open\id_open.cpp.o
     beacon->oui_type = 0x13;
           ^`
I was able to fix it by modifying the `ieee80211_beacon struct ` in `odid_wifi.hadding` adding the missing variables.

`struct __attribute__((__packed__)) ieee80211_beacon {
    uint64_t timestamp;
    uint64_t element_id;
    uint64_t length;
    uint64_t oui_type;
    uint64_t oui;
    uint16_t beacon_interval;
    uint16_t capability;
};`

I just wanted to let you know and thank you for your work.

sxjack, 2021-11-04:
You shouldn't be compiling anything in lib\libopendroneid. 

From the README -

"Needs opendroneid.c, opendroneid.h and odid_wifi.h from opendroneid. The wifi.c in this directory is slightly modified from the one in opendroneid."

The last time that I checked for compatibility with the opendroneid library was a few weeks ago when they issued release 1.0.

pierreSCpapon, 2021-11-05:
Thank you for your reply.

"Needs opendroneid.c, opendroneid.h and odid_wifi.h from opendroneid" 
Your program has dependencies with these files so I included them as a library and replaced wifi.c with your version.

However, it fails to compile as your version of `wifi.c` refers to `element _id` in the ` ieee80211_beacon` struct

`error: 'struct ieee80211_beacon' has no member named 'element_id'`

I was able to fix it by modifying the ieee80211_beacon struct in odid_wifi.h adding the missing variables.

I hope this makes sense!

friissoren, 2021-11-05:
I think I know what the problem is. The ESP32 project has its own local copy of wifi.c (with slight modifications). However, that file was changed recently in the core-c project to reflect some changes in the structure definitions in opendroneid.h (for adding Wi-Fi Beacon support) and the ESP32 project therefore no longer compiles correctly if you pick up the latest opendroneid.h from core-c.

wifi.c should be updated in the ESP32 project.

Probably a better long term solution would be figuring out what exactly those differences in that file are and get them committed to core-c somehow (possibly under a compile flag that identify the ESP32 environment). Then the entire core-c project could be added as a git submodule project in the ESP32 project and they wouldn't run out of sync. That would also get rid of the need to manually copying files.

sxjack, 2021-11-05:
@friissoren The difference is a few #if defined()'s so that it compiles on systems that do not have a MONOTONIC clock. I'll try to whistle up the enthusiasm to do a pull request.

sxjack, 2021-11-05:
@pierreSCpapon wifi.c is now compatible with the latest opendroneid code.

pierreSCpapon, 2021-11-08:
Compiles without a problem now! Thank you.

closed 2021-11-08 by pierreSCpapon