sxjack/uav_electronic_ids #12: ESP32 clears WiFi timestamp

sxjack, 2021-11-19:
The ESP32 function esp_wifi_80211_tx() sets the timestamp field to zero.

simon88, 2022-02-03:
@sxjack there is a fix for that ?

sxjack, 2022-02-03:
I couldn't find one. It seems to be feature of espressif's api. It is mentioned in one of the open source French ID programs (emetteur_balise.ino, there is a comment that says 'overwritten by hardware').

simon88, 2022-02-03:
This is strange because I've just tried it with ESP32 And it's works, I've set hours minutes seconds in utm_data and I've it in the packet in WireShark @sxjack

sxjack, 2022-02-03:
I'll have to have a look at the version information on my collection of ESP32's and have a go with the newest.

It occurs to me that we may be talking at cross purposes here. The timestamp that I am talking about is the WiFi one not the one in the opendrone ID frame.

closed 2022-05-27 by sxjack