sxjack/uav_electronic_ids #16: esp32-s3 support discussion.

davidbuzz, 2022-08-04:
I just wanted to note for others that ESP32-S3 works under the following conditions:

Arduino IDE -> File Menu -> Preferences -> "Additional Board Manager URLs:" https://github.com/espressif/arduino-esp32/releases/download/2.0.3-RC1/package_esp32_dev_index.json
Arduino IDE -> Tools Menu -> Board -> Boards Manager -> [search for 'esp32'] ->Select Version [drop-down]-> 2.0.3-RC1 -> Install 
 Arduino IDE -> Tools Menu -> Board -> "ESP32 Arduino" ->"ESP32S3 Dev Module" [choose it]
In your file manager navigate to ~/Arduino/libraries/  folder 
in this folder you need a folder called 'id_open' and a folder called 'utm'  that come from https://github.com/sxjack/uav_electronic_ids.  i cloned that repo elsewhere and symlinked these two folders into ~/Arduino/libraries, but how they get here doesn't matter too much.
You can then open  one of the 'example/s from id_open eg:
Arduino IDE -> File Menu -> Examples ->[scroll] Examples from Custom Libraries -> id_open -> random_flight
[plugin your ep32-s3 with usb cable using the usb port labeled "USB" on the pcb]
Arduino IDE -> Tools Menu -> Port:... -> /dev/ttyACM0
Press 'Upload' '"arrow" in IDE green bar. 
If board does not flash, hold-down BOOT pushbutton on pcb while pressing RESET pushbutton briefly [to force it into bootloader mode] and retry.
done, ESP32-S3 is now running and emitting test/demo remote-id bluetooth messages.

davidbuzz, 2022-08-04:
its probably works with anything newer as well, but i haven't tried that, for example 2.0.4 has come out since i did this.

davidbuzz, 2022-08-06:
We'd like to work on adding Bluetooth 5 support with -S3 hardware asap, @sxjack are you around ?  would like to colaborate.

sxjack, 2022-08-08:
I am unlikely to do any further work on this unless the UK government decides to go for direct remote ID and, at the moment, the signs are that they are not going to.

I addition, I don't have anything that supports BT5 and if I were to start looking at it, it would probably be with Nordic hardware.

davidbuzz, 2022-08-08:
Would it help if i posted u an esp32-s3, or two,  which definitely has BT5 support in hardware?

davidbuzz, 2022-08-08:
https://www.espressif.com/en/products/socs/esp32-s3

sxjack, 2022-08-11:
> Would it help if i posted u an esp32-s3

No

davidbuzz, 2022-08-19:
For those looking for RemoteID on esp32-s3 or esp32-c3 , with BT4 and BT5 and WiFi NaN support... weve written code that does all of those things at: https://github.com/ArduPilot/ArduRemoteID

closed 2022-08-20 by sxjack