sxjack/uav_electronic_ids #18: reboot after flashing

kirbizz, 2022-12-07:
Hello, download last commits, built the project, upload and my esp-32-wroom falls in cyclic reboot? What i am doing wrong?

kirbizz, 2022-12-08:
I am using random_flight, the problem is somewhere in squitter.init(&utm_parameters) func. Yesterday It started working, but today stopped and i have the same issue. Can you send me the screen of the board settings in tools? Frequencies, Partition scheme

sxjack, 2022-12-08:
The only thing that I change from the defaults is the partition scheme which needs to be 1.9+ MB if you are using WiFi and BLE. CPU freq is 240 MHz, flash 80 MHz & 4 MB.

https://github.com/sxjack/uav_electronic_ids/blob/main/ENV.md

The other thing has caused rebooting for me is using Serial1.

sxjack, 2022-12-08:
Have you tried turning off the Bluetooth and/or Wifi?

kirbizz, 2022-12-08:
thanks, the problem is in wifi_beacon enabled workmode
if i disable it, everything works, don't you know what could be the problem?

sxjack, 2022-12-08:
If you disable Bluetooth does WiFi work?

There are two options for how the WiFi works that you could try changing.

1) USE_BEACON_FUNC  (in id_open.h) determines whether the beacon packets come from opendroneid's code in wifi.c or mine. This was done for a couple of reasons, to investigate why Android seems to ignore lots of beacon packets, mine are closer to what an ESP32 normally puts out (Android still ignores lot of them) and because Arduino Pi Pico W hates wifi.c.

2) About half way down id_open_esp32.cpp there is an #if that determines whether the program uses the WiFi class or calls the esp_ functions directly. It is currently set to do the latter because the former means that the ESP will transmit it's own beacon packets as well as ours.

kirbizz, 2022-12-08:
1. Bluetooth enable works perfect with BT
#define ID_OD_WIFI_NAN    0
#define ID_OD_WIFI_BEACON 0
#define ID_OD_BT          1 
#define USE_BEACON_FUNC   0

2. WiFi enable works , no crash, but we can't see it in mobile phone (my of course)
#define ID_OD_WIFI_NAN    0
#define ID_OD_WIFI_BEACON 1
#define ID_OD_BT          0 
#define USE_BEACON_FUNC   0

3. WiFi enable works , no crash, but we can't see it in mobile phone (my of course)
#define ID_OD_WIFI_NAN    0
#define ID_OD_WIFI_BEACON 1
#define ID_OD_BT          0 
#define USE_BEACON_FUNC   1

4. WIFI + BT enable, crashes
#define ID_OD_WIFI_NAN    0
#define ID_OD_WIFI_BEACON 1
#define ID_OD_BT          1 
#define USE_BEACON_FUNC   1

5. WIFI + BT enable, crashes
#define ID_OD_WIFI_NAN    0
#define ID_OD_WIFI_BEACON 1
#define ID_OD_BT          1 
#define USE_BEACON_FUNC   0

sxjack, 2022-12-08:
So you can have BT or WiFi. I may go through my collection of ESP32's later and see if any of them do the same. The two currently on my bench don't.

Android or Apple? For Android you can go into the developer options and turn off beacon throttling and up the rate in id_open.h, but I've never got it to work anything like as well as BT4. My unix and ESP32 scanners see the packets just fine.

kirbizz, 2022-12-08:
The thread about this, that won't work simultaneously
https://esp32.com/viewtopic.php?f=13&t=6707&start=60

Tuburni, 2023-03-15:
Hello

@sxjack, can you help me with these things? 

C:\Users\User\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.cpp:295:6: error: no declaration matches 'void ID_OpenDrone::set_self_id(char*)'
 void ID_OpenDrone::set_self_id(char *self_id) {
      ^~~~~~~~~~~~
C:\Users\User\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.cpp:295:6: note: no functions named 'void ID_OpenDrone::set_self_id(char*)'
In file included from C:\Users\2683126\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.cpp:54:
C:\Users\User\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.h:104:7: note: 'class ID_OpenDrone' defined here
 class ID_OpenDrone {
       ^~~~~~~~~~~~

exit status 1

Compilation error: no declaration matches 'void ID_OpenDrone::set_self_id(char*)'

closed 2023-03-16 by sxjack