sxjack/uav_electronic_ids #19: Some errors that i don't understood

Tuburni, 2023-03-15:
Hello

@sxjack, can you help me with these things?

C:\Users\User\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.cpp:295:6: error: no declaration matches 'void ID_OpenDrone::set_self_id(char*)'
void ID_OpenDrone::set_self_id(char self_id) {
^~~~~~~~~~~~
C:\Users\User\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.cpp:295:6: note: no functions named 'void ID_OpenDrone::set_self_id(char)'
In file included from C:\Users\2683126\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.cpp:54:
C:\Users\User\Desktop\uav_electronic_ids-main\uav_electronic_ids-main\id_open\examples\random_flight\id_open.h:104:7: note: 'class ID_OpenDrone' defined here
class ID_OpenDrone {
^~~~~~~~~~~~

exit status 1

Compilation error: no declaration matches 'void ID_OpenDrone::set_self_id(char*)'

sxjack, 2023-03-15:
It looks like I forgot to update a file. I'll fix it later this evening.

Tuburni, 2023-03-15:
@sxjack

Thanks a lot
I will wait a lot
I need this for my graduation project

3-Hydroxytyramine, 2023-03-15:
Hello, I have the same problem. I will wait for the fix

sxjack, 2023-03-15:
Fixed.

sxjack, 2023-03-15:
If you are studying C++, you really should have realised that it was a missing declaration in the h file.... :-)

Tuburni, 2023-03-16:
Hello @sxjack

Thank you very much, everything works)))
Sorry I'm a noob in  C++)

closed 2023-03-16 by sxjack