sxjack/uav_electronic_ids #20: id_scanner wifi channel

michaelzing, 2023-04-24:
I see the wifi channel that the ESP32 scans is set to channel 6 in id_scanner.ino. I'm wondering why channel 6 was selected specifically? The ASTM standard allows wifi beacon packets to be broadcast on many other channels as well, right? 

From the ASTM standard: "The broadcast message shall (BWFB0030) be transmitted on either: ... Any Wi-Fi channel of the 2400 MHz to 2483 MHz band ... Or Any Wi-Fi channel of the 5150 MHz to 5895 MHz band".

sxjack, 2023-04-24:
It looks for F3411 WiFi Beacon, NAN and French beacons.

michaelzing, 2023-04-26:
Got it, thank you. I see that in id_scanner.ino, line 326 sets the wifi channel to channel 6:

`esp_wifi_set_channel(6,WIFI_SECOND_CHAN_NONE);`

Is it possible to change this to scan on another channel in the 2.4 GHz band (i.e. any channel between 1 and 11)?

friissoren, 2023-04-26:
I will let someone else comment on the functionality of the code, but from a standards perspective, please see the answer [here](https://github.com/sxjack/uav_electronic_ids/issues/21#issuecomment-1522838970).

michaelzing, 2023-04-26:
thank you, i would still like to see if anyone can comment on the functionality of the code.

sxjack, 2023-04-26:
> Is it possible to change this to scan on another channel in the 2.4 GHz band (i.e. any channel between 1 and 11)?

Obviously you can change it to whatever you want and/or set the second channel, but if you do, see the comment above the esp_wifi_set_channel call.

closed 2023-04-27 by michaelzing