sxjack/uav_electronic_ids #21: id_open random_flight.ino broadcasting on a different wifi channel

michaelzing, 2023-04-26:
I see in id_open.h, that WIFI_CHANNEL is set to 6. Is it possible to change this channel to any other channel in the 2.4 GHz band (specifically, any channel between 1 and 11)?

friissoren, 2023-04-26:
I assume you are asking about Wi-Fi Beacon, since Wi-Fi NaN must always be on channel 6 or 149.

I will let someone else comment on the functionality of the code but both the ASTM F3411-22a and the ASD-STAN FprEN_4709-002:2021 standards define that if you use channel 6 or channel 149 (in the 5GHz band) for Wi-Fi Beacon transmission, the messages must be sent at least once per second (or per third second for some message types, but since you have to use message packs for Wi-Fi Beacon, those message types effectively end up being transmitted also once per second). 

If you use any other channel in the 2.4 GHz or the 5 GHz bands, the message transmit interval must be less than 200 TU (Time Unit, 1 TU = 1024 microsecond). The reason is to increase the chance that a receiver picks up the messages. For channel 6 the receiver only needs to scan a single channel but if any other channel is used, the receiver needs to cycle through all channels and can only listen on each one for a limited time.

michaelzing, 2023-04-26:
Hi, thank you so much for the response, that makes a lot of sense. Can anyone still comment on the functionality of the code? I'm using the transmitter in order to test a receiver that I'm working on, that ideally can receive on any channel.

sxjack, 2023-04-26:
> Is it possible to change this channel to any other channel in the 2.4 GHz band (specifically, any channel between 1 and 11)?

You can change it to whatever you want, but as Soren says, if you move away from 6, you should increase the rate (change BEACON_INTERVAL). 

If you are going to be using an Android phone as a receiver, it's probably worth increasing the rate anyway. I've played around with the rate a bit over the last couple of years, IIRC the current default in the code is higher than the standard requires.