sxjack/uav_electronic_ids #22: National IDs, WIFI Settings and FAA Approval

Greg-C10, 2023-05-04:
I was looking at the UAV Electronic IDs code (buckminster example).  I was wondering if National ID and the WIFI settings should be dynamically configured based on GPS location?  Seems like it would increase memory requirements to store country configs.  Will this have to be reconfigured and compiled if someone travels across countries?

What are your concerns about getting it certified in the US for FAA approval?  What needs to be added?

Thanks

sxjack, 2023-05-04:
I am not sure that the national settings are worth a lot of effort. I probably should have a #define or two in id_open.h to set them with a default of ZZ. IIRC country was one of the thing that I played with (along with most of the stuff in id_open_beacon.cpp) to see if it would improve reception using Android.

I can't see how an open source ID could meet the FAA requirements.

friissoren, 2023-05-05:
I am not completely sure but I would expect that some variations of the [ArduRemoteID](https://github.com/opendroneid/opendroneid-core-c#transmitter-examples) implementations have been FAA certified. E.g. the Bluemark products. But please research this yourself a bit more carefully.

sxjack, 2023-05-05:
I really don't understand how a device like the db122fpv which takes an external GPS feed got on the approved list. It would be pretty easy to have a GPS feed that never went above 400ft.