sxjack/uav_electronic_ids #24: reboot loop with id_open code

hsuq510, 2023-05-26:
Hi there, 

I was success with running france example and scanner as well. However I have got a reboot loop when trying to run the finland code and any other open_id example with the following stack trace. Tried on two different ESP32 board, both ESP32-S and one ESP32-WROOM-32. Im using board "Esp32 Dev Module" included in library esp 1.0.6 with Arduino IDE 1.8.9 with the following params:

240MHZ Wifi/BT, Flash Mode QIO, Flash size 32Mb, Partition scheme 2MB/NO OTA/2MB SPIFF, PSRAM disabled.

Decoding stack results
0x4008f088: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008f305: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x4019ed64: esp_phy_rf_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c line 197
0x4019efce: esp_phy_load_cal_and_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c line 673
0x4010ff4f: esp_bt_controller_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c line 1528
0x400d6106: btStart at /Users/syn/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/esp32-hal-bt.c line 45
0x400d37f5: BLEDevice::init(std::__cxx11::basic_string, std::allocator >) at /Users/syn/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/BLE/src/BLEDevice.cpp line 339
0x400d142d: init2(char*, int, unsigned char*, unsigned char) at /Users/syn/Documents/Arduino/libraries/id_open/id_open_esp32.cpp line 212
0x400d0b0c: ID_OpenDrone::init(UTM_parameters*) at /Users/syn/Documents/Arduino/libraries/id_open/id_open.cpp line 253
0x400d081d: setup() at /var/folders/pp/t4fyr3yn03957kjjm25g393w0000gn/T/arduino_modified_sketch_51192/finland.ino line 27
0x400d6402: loopTask(void*) at /Users/syn/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp line 18
0x4009037a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

sxjack, 2023-05-26:
Are you trying to transmit both WiFi and Bluetooth? If so, see issue #18 and the last paragraph of the id_ipen readme.

hsuq510, 2023-05-26:
Yes, you can close this same as #18. Thanks

hsuq510, 2023-05-26:
Close

closed 2023-05-26 by hsuq510