sxjack/uav_electronic_ids #25: random_flight.ino: Non-RID Wi-Fi packets being broadcast

Michaelszeng, 2023-06-12:
I'm running random_flight.ino on an ESP32 using PlatformIO. Sniffing the wifi packets using Wireshark, I am finding that valid Remote ID packets are indeed being broadcast and properly decoded, but, strangely, Wireshark also detects other wifi packets broadcast by the ESP32 that aren't OpendroneID packets. The non-remote ID packets are sent by a mac address that is incremented by 1 compared to the mac address that is broadcasting the valid remote ID packets. The hex dump from the non-remote ID packets looks completely different.

Here is a screenshot from Wireshark, where you can see both the valid Remote ID and invalid packets broadcast by the ESP32:
[image]

And here is the hex dump of those invalid packets broadcast by the ESP32:
[image]

Any idea why it is doing this? Maybe it is responding to probe requests? How can I disable this?

sxjack, 2023-06-12:
Sounds like the ESP32's WiFi stack is sending its own frames.

The ESP32_WIFI_OPTION flag controls how the WiFi is opened in id_open_esp32.c and IIRC setting it to 0 is supposed to solve the extra frames problem, but I have never run it through platformio.

Michaelszeng, 2023-06-13:
I've re-tried in the Arduino IDE too... just cloned the repo, moved all the relevant files (from the id_open folder, utm folder, and opendroneid files) into the random_flight folder, and uploaded random_flight.ino the ESP32. But the issue persists.

Maybe it's that I'm using an ESP32 Pico Kit? I've also tried on an ESP32-C3-mini-1u kit, but got the same issue.

Do u possibly have any other ideas?

I appreciate your help though, thank you.

sxjack, 2023-06-13:
My ESP32's are a few years old and are mostly WROOMs and IDF v3. There is another issue that indicates that there are differences between the firmware on the units that I have and newer ones.

I have run up a TTGO 32 (?) and a WROOM. I see remote ID frames and probe responses on from the TTGO and only remote ID frames from the WROOM.